daid/LegacyCura #1446: Cura Won't Start - Windows 8

bstott, 2016-04-08:
Cura will not start. I get a splash screen. Then nothing. When the splash screen is clicked on it goes away. I have uninstalled the previous versions with the uninstall.exe and Windows program uninstall.

Application Version: 15.04.03

Platform: Windows 8.0

Qt: 4.8.5
PyQt: 4.10.3
Python: 2.7.8
Display Driver: AMD Radeon HD 6310 Graphics, Driver version 14.100.0.0

Steps to Reproduce:
Try to run Cura 15.04.03

Actual Results:
Splash Screen stays on window until clicked on. Then disappears without Cura application starting.

Expected results:
Should be able to see the Cura GUI and use the application to slice stl files and 3D print.

Additional Information:
N/A

Note: Pronterface (python app) works.

Output log attached. Note that the output log is located in the ...User/.cura directory not in Appdata...

[output_log.txt](https://github.com/daid/Cura/files/210650/output_log.txt)

daid, 2016-04-08:
We've seen this problem a few more times. We're not 100% sure what is causing it, it has to do with initialization of the 3D rendering that is failing.

Re-installing your 3D drivers sometimes helps.

We're also launching the beta of our new Cura 2.0 soon, that might fix this problem for you as well.

bstott, 2016-04-08:
Updated to Cura 5.04.05 on April 8, 2016.
Attached newest output_log.txt below.
[output_log.txt](https://github.com/daid/Cura/files/210948/output_log.txt)

bstott, 2016-04-08:
Oops! Closed by accident ---- Problem still exists with Cura 15.04.05

bstott, 2016-04-08:
Glut or FreeGlut?

Curious. I read in the output log that there is, "TypeError: 'NoneType' object is not callable" after a each call for a GLUT function/feature ---- > Could the developers for CURA use FreeGlut instead? Or is it being used?