daid/LegacyCura #1448: Rotation does not work

AlexanderManes, 2016-04-12:
When using a mac's trackpad, the rotation function does not work when I right click or control click. I am not sure if this happens also with a mouse.

daid, 2016-04-13:
This is a known issue, it's due to the fact that wxWidgets is sometimes incorrectly detecting the shift key being pressed and hold down.

echox, 2016-04-18:
Is there any workaround?

I mean, its not that it doesn't work sometimes, it is impossible with an actual version of Cura to rotate the view. It happens of course also with a mouse.
Right click movement works and zooming with the wheel.

daid, 2016-04-18:
Not that I know off.
We have our 2.1 beta out now. So we're focusing on that. It's no longer using wxWidgets, as we had too many bugs with it (like this one)

echox, 2016-04-18:
I wasn't aware that there is a new 2.1 beta.
Thanks. Tried it and of course rotation is working there.

So I assume support and development will move over to https://github.com/Ultimaker/Cura over time?

nallath, 2016-04-18:
It already moved there :)

daid, 2016-04-18:
It's moved there for 2.1 and onwards. This is just for old 15.04 support.

Note, I'm closing this and advising everyone with OSX to move to 2.1 beta. As it solves a whole heap of problems on OSX.

closed 2016-04-18 by daid