daid/LegacyCura #1449: Issues with model slicing

jtkdaddy, 2016-04-14:
I have a part attached that refuses to slice.  It will complete loading successfully and displayed in the main window, but the progress bar to slice it never shows up.  Also, switching to layer view displays nothing and am unable to generate a gcode file.

[Model.zip](https://github.com/daid/Cura/files/218996/Model.zip)

daid, 2016-04-14:
Which version did you use? This model works fine here.

jtkdaddy, 2016-04-14:
15.04.4

daid, 2016-04-14:
Same version as I've used. Could it be that you have anti-virus or firewall software that is causing problems?

XiaoShiZi, 2016-04-14:
15.04.3 Gives no Problem to genarate layers and toolpath.
Greetz XiaoShiZi

XiaoShiZi, 2016-04-14:
15.04.5 on Ubuntu 32 bits gives good results to.

Greetz XiaoShiZi

2016-04-14 22:42 GMT+02:00 Xiao Shi Zi 1 <xiaoshizi20@gmail.com>:

jtkdaddy, 2016-04-14:
Installed 15.04.5 which seems to fix it for me.

jtkdaddy, 2016-04-14:
Looks like it actually didn't work.  I opened a different model and was able to generate the gcode.  When I opened the problematic model attached to this thread, the save gcode icon remained enabled but Cura actually generated the gcode for the first model that I opened.  I restarted Cura and opened the problematic model first and now the save gcode icon is disabled.