daid/LegacyCura #1451: self level progam stops

Sneedr, 2016-04-16:
When I use the Run bed leveling wizard it stops and stalls out at the going to rear middle point and stops there like it cant find it. Could someone help me with it. I can print the test and it comes out okay but all the others I have made do not work and it tries to print off the table.

BagelOrb, 2016-04-18:
You should try out the beta of the new Cura; the download link is just
below the normal download link on software.Ultimaker.com

(Report it's problems on github.com/Ultimaker/cura)

Sneedr, 2016-04-19:
I should  tell you about the  printer I have it is not an Ultimaker. It is a cheap copy from China. I am wondering if I load your  firmware upgrade if that would help.  The put it up in ebay as a Prusa I3. Your beta does not find my machine connected so it won’t work for me. If you have any adivice I would be thrilled to hear some so I can make this print like it is suppose to. I does print the test print out just fine. I am using Sketch up for my design but it still prints off the the build table, when I use anything I made from sketch up.

Thank you

Ghostkeeper, 2016-04-20:
If it is a Prusa I3, did you try loading the latest Prusa firmware? You can get them from their site: http://www.prusa3d.com/drivers/

Sneedr, 2016-04-20:
kb0oqz@outlook.com has shared a OneDrive file with you. To view it, click the link below.

[icon]

printer stats.docx<https://onedrive.live.com/redir?page=view&resid=ACE1E20D6D58C082!1388&authkey=!APy0e0JDFtNtx40>

Okay I am going  to try  and put  a link to the printer or at least a picture of the machine and what it says about it. If you know this printer or have seen it before could you tell me it’s real name and anything that could help me.