daid/LegacyCura #1453: Switching between multiple different printers causes slicing errors

N720, 2016-04-20:
Switching between printers in the Machine menu appears to cause problems with slicing.  The attached image is of a gcode file re-imported into Cura after Cura sliced it.  Printer profiles I have are Printrbot Simple Metal, Printrbot Metal Plus, and Witbox.  It has happened on gcode exports for both the Witbox and the Printrbot Simple Metal. 

Also attached is the gcode file as txt

[image]
[NeckV1_gcode.txt](https://github.com/daid/Cura/files/227039/NeckV1_gcode.txt)

daid, 2016-04-20:
This is a different bug, the gcode is fine (load it in a different viewer). The bug is that the loader and the viewer are not properly synced causing problems that not everything is rendered.

N720, 2016-04-21:
Thank you for checking out the gcode.  Here's an example of how the prints turn out.  My process is to export gcode from Cura, import to Octoprint, and print.  Must be a problem with Octoprint.

[image]

daid, 2016-04-21:
Looks more like your printer cannot keep up with the extrusion rates Cura is asking.

closed 2016-04-21 by daid

N720, 2016-04-22:
I believe I located one of the problems.  When I re-installed Cura it had the filament size set to 2.7 mm instead of 1.75.  Thank you for your help in fixing the problem.