daid/LegacyCura #1455: Slicing fails silently on 16.04 - new versions will not install

gazhay, 2016-04-23:
16.04 ships with a newer python3 than cura depends on so will not install.

On my existing install of cura no slicing works at all. I can load models but nothing happens. 
"Show Slice Engine Log...." does nothing.

daid, 2016-04-23:
There is no 16.04 version of Cura.

closed 2016-04-23 by daid

gazhay, 2016-04-23:
16.04 of Ubuntu, released yesterday, hence "ships with"

daid, 2016-04-23:
I'm not a mind reader. Nowhere you tell you use linux, nowhere you tell you use ubuntu. Nowhere you tell which version of Cura you tried.

Pretty much no information is provided. So no help is provided.

gazhay, 2016-04-23:
I appreciate it was a poor bug report.

BagelOrb, 2016-04-23:
So what version of Cura are you trying?

Try the Cura 2.1 beta version!
Log bugs for that software under Ultimaker.Cura instead of daid/Cura

gazhay, 2016-04-23:
I have logged in the correct repo now.

Fwiw I tried to install 2.1 Beta and 15.06.03, both fail as described on python.
If you force them to install they crash on start.