daid/LegacyCura #1456: The "spiralize the outer contour" option generates a hollow model

Yansie6, 2016-04-23:
In the Expert Configs, underneath "Black Magic": When the "Spiralize the outer contour" option is checked, Cura does not generate any filling. It does not matter what the infilling percentage is set to.

Cura version 15.04.5

Ghostkeeper, 2016-04-25:
This is intended. Spiralize Outer Contour *only* draws the contour of the model. No infill. No support. Nothing. It's useful for printing things like vases and mugs.

closed 2016-04-29 by BagelOrb

mrlunk, 2017-05-25:
How do I set the wall thickness for this ?
It is showing only 1 outer layer in CURA...

BagelOrb, 2017-05-26:
Spiralize can only work with one contour. The whole idea is that you
gradually move up while printing. That only makes sense when the is a
single contour.

kyrreaa, 2017-05-26:
But if you set external perimeter extrusion width you can control wall
thickness.