daid/LegacyCura #1457: im useing an pursa i3 diy printer and cura isint connecting to printer

ghost014, 2016-04-24:
i carnt print online as the option to online print dosent pop up

nallath, 2016-04-25:
Print online? Cura doesn't have any support to print online.

closed 2016-04-25 by nallath

ghost014, 2016-04-25:
Online as in connect printer to laptop

Ghostkeeper, 2016-04-25:
There is no option to print via Wifi either. USB might sorta work since it is both Marlin (but you might run into some strange bugs).

ghost014, 2016-04-25:
usb dosent work as it doesn't pop up with online-printing window and when I try and print from sd card is dosent work either

Ghostkeeper, 2016-04-25:
>print from sd card is dosent work either

Ah, then likely either the machine definition is broken in Cura (something with start and end g-code, how it needs to warm up the nozzle, start the fans, etc.) or the printer itself is broken.

ghost014, 2016-04-25:
ah cheers, anything I can do to fix it?

Ghostkeeper, 2016-04-25:
Depends. What happens when you put an SD card in the printer with g-code from Cura and start the print? Does it do anything? Move the head? Warm the nozzle? If it does do something but not everything, e.g. it forgets to heat the nozzle, then it might accidentally be omitted from the g-code. OR it might just be that the heater in your printer is broken! So it's hard to say.

In any case, perhaps you could open the .gcode file with a text editor and copy the first bit here, up until and including the first "G1" line? That'll help us determine if the heat command is forgotten or not.

ghost014, 2016-04-26:
M190 S70.000000
M109 S210.000000
;Sliced at: Tue 26-04-2016 23:12:15
;Basic settings: Layer height: 0.15 Walls: 2 Fill: 10
;Print time: 4 minutes
;Filament used: 0.116m 0.0g
;Filament cost: None
;M190 S70 ;Uncomment to add your own bed temperature line
;M109 S210 ;Uncomment to add your own temperature line
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F9000
;Put printing message on LCD screen
M117 Printing...

;Layer count: 66
;LAYER:0
M107

ghost014, 2016-04-26:
this is just printing a very basic 10mm cube also the printer heater/ bed heats up but cuts out half way though

Ghostkeeper, 2016-04-28:
The g-code that Cura provides looks fine. I strongly suspect that something is up with the printer. Especially if it apparently stops heating halfway through a heat command.

ghost014, 2016-04-28:
ok cheers, my friend recons its the driver software but I carnt find it anywhere

Ghostkeeper, 2016-04-29:
Driver firmware for your printer can be downloaded here: http://www.prusa3d.com/drivers/

But you mentioned you have a DIY printer, so it might be slightly different?

ghost014, 2016-04-29:
Its a pursa i3 diy printer

Ghostkeeper, 2016-05-02:
Perhaps this helps: http://prusa3d.com/downloads/firmware/EN-how-to-choose-firmware.pdf