daid/LegacyCura #1458: Mirrored Text will not printed as it was created in Cura 15.4.2

Bombastino, 2016-04-24:
[image]
[image]

Hi,
I have a problem with Cura. I designed a Cookie-Cutter with Text. The Text was mirrored and that is what cura shows. But the printing Result was that the Text was not mirrored. 
What I have to do to correct. 
Thank you for your help

BagelOrb, 2016-04-24:
Try the 2.1 beta. The download link is right below the download link of
15.04.

Ghostkeeper, 2016-04-25:
Weird. Something is flipping the Y-axis. What happens when you activate layer view, or when you load the g-code in another application like Repetier-Host?

If it looks properly mirrored there, I'd suspect something is up with the printer.