daid/LegacyCura #1462: First layer extrusion width in Cura 2.1 beta

elliotf, 2016-05-04:
Hello and thank you very much for Cura!

#769 (first layer extrusion width) at %150 of normal width and printing the first layer slowly (~30mm/s) solves my first layer adhesion problems.

I'm trying out 2.1 and it looks quite nice, but I'm having a hard time finding the first layer extrusion width setting.  Was that removed or was it renamed to something else?

Thank you!

nallath, 2016-05-04:
2.1 issues should be reported in https://github.com/Ultimaker/cura

BagelOrb, 2016-05-04:
It was removed, due to the fact that we now have extrusion width settings per feature type (skin, walls, infill etc.)
Making a first layer extrusion width setting per feature type would generate too many settings...

elliotf, 2016-05-04:
Thank you for the information and I apologize for the noise..   I'll take the conversation over to Ultimaker/cura and continue the conversation there.  Thank you!

closed 2016-05-04 by elliotf