daid/LegacyCura #1467: No temperature sended to the printer in the Gcode

SteenJ, 2016-05-10:
A couple of times in the week, I generate Gcodes that have no temperatures in the Gcode. The 3D printer started with warming. And then I check what’s the setting of the temperature, I found out that in some cases these is stated as 0 degrees. 

I would like to see that I can check this and maybe edit these before generating the Gcode. In some cases it could be useful to edit the default temperature. But at this moment this can only be changed in the printer, but not in the Gcode settings. So it is harder for me to make 5 times the same product, because than in need to edit it 5 times instead of doing this before generating.

Ghostkeeper, 2016-05-11:
If the g-code flavour is UltiGCode, there should be no temperature commands in the g-code (UM2 machines don't read temperature commands but only allow setting the temperature on the machine).

For any other machine it should include temperature commands in the g-code. Though some (I don't know offhand which ones) might define some temperature in the starting g-code.

SteenJ, 2016-05-11:
OK, in this specific case it is the Ultimaker 2 Extended+. But I have often that there is no temperature installed and he want to start with a nozzle of zero degrees. But as I understand it right, the problem isn’t in Cura but in the design of Ultimaker. Maybe I can import some new materials to solve this problem…

BagelOrb, 2016-05-11:
If you accidentally sliced for an Ultimaker 2 type of machine, but you use
that code on an Ultimaker original, then the printer will try to print
without warming up.

Does that explain what is happening to you?