daid/LegacyCura #1473: New MAC UPDATE 15.04.6

sscoban, 2016-05-18:
There are lost of issues with this update;

*  Homing Z0 for some reason is reversed when I use;
G28 X0 Y0
G28 Z0
Z axis goes other way and table crashes the bottom
I didn't have this issue with previous version, by only using G28 solves this issue.

*  After I save g-code to SD card, notification bar appears but there is nothing written on it and no safe remove etc. 

*  After print reaches few layers, z axis raises for no reason and crashes the print thus destroys it.

!!! By saying all that, I know that there is nothing wrong with my printer, I just printed couple of parts with older version few hours ago.

daid, 2016-05-19:
Which printer do you have?

GrimbixCode, 2016-06-01:
> After print reaches few layers, z axis raises for no reason and crashes the print thus destroys it.

I got the same problems with the Ultimaker 2.