daid/LegacyCura #1478: Cura dual extruder thin walls not generating gcode

jaredabrandt001, 2016-05-24:
Application Version: Cura 15.04.6 or Cura engine running in Repetier Host
Platform: Windows 7 x64
Qt: don't know
PyQt: don't know
Display Driver: Quadro 2000 version 341.44
Printer: Large format Reprap Mendel

Steps to Reproduce:
Model size is approx 24"x24"x6" (large print), dual color. Files are imported and extruders assigned to different models. This machine has printed many large files, and I've used cura to print many single extruder prints. I'd like to use Cura for dual extruder prints also and get away from Slic3r, but at the moment Cura does not process the gcode properly. Slic3r does, but it takes 4-8 hours to slice.
Problem occurs in Cura 15.04.6 and Repetier host the same.
Cutting the model in half (reducing the file sizes) does not help.

Actual Results:
After slicing, gcode shows missing walls. Missing walls tend to be in the same location. Part prints with missing walls. Wall thickness is set to 1.25mm with a 0.6mm extruder. Perimeters are set to 0.6mm. Mesh quality does not significantly impact the issue.
Note: If all models are set to the same extruder (no dual extrusion) model gcode is generated correctly, so this issue is only with more than 1 extruder.

Expected results:
Should generate gcode where the model is solid.
[image]
[image]

Any help would be greatly appreciated!
Jared

BagelOrb, 2016-05-25:
We are not developing on the Cura 15 anymore.
Cura 2.2 will most probably support dual extrusion and there the bug will
be solved.

jaredabrandt001, 2016-05-25:
Is there any way in the meantime I can understand why and when this will happen...other than just avoiding Cura when walls under 4x's the nozzle width and dual extruders are involved?

jaredabrandt001, 2016-05-25:
To be fair, Simplify3D comes nowhere close to working with models like this either, so I'm forced to use Slic3r.
-I also find Slic3r's dual extruder temperature management handy.  I've put the gcode in for Cura and models like this print almost perfect by simply turning the inactive extruder way down.  Waaay better than an ooze shield, and faster on larger models (and less waste).  Would be something to consider adding to Cura 2.2?

BagelOrb, 2016-05-25:
I don't know why that happens and I don't think there is a work around, but
maybe @daid can inform us.

I don't know how the temperature management in slic3r works exactly, but
the new Cura will have a feature which automatically turns the inactive
nozzle down to the standby temperature and start heading up in time.

Ooze shields and prime towers don't seem to beneficial when you tuned your
print profile right.