daid/LegacyCura #1483: No port in Linux

GettingNowhere, 2016-05-26:
If I try and print on my Makerbot dual extruder machine using /dev/ttyACM0 at any fixed baud rate, it says "Connecting" for a while, then eventually just says "Closed"
If I use "Auto", it just says "Error No more baudrates to test and no s"
I cannot extend the box to see any more of the sentence.

Installed:
Ubuntu 16.04
Cura 15.04.5
I installed Cura 15.04.5 as well as
pypy
python-gpgme
python-opengl
python-numpy
python-serial
python-setuptools
curl
cura-engine
and arduino
as suggested by some website or other.
` dmesg | tail -n15
[ 1257.755999] usb 5-1: new full-speed USB device number 4 using ohci-pci
[ 1257.933224] usb 5-1: New USB device found, idVendor=23c1, idProduct=d314
[ 1257.933238] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[ 1257.933246] usb 5-1: Product: The Replicator
[ 1257.933252] usb 5-1: Manufacturer: MakerBot Industries
[ 1257.933257] usb 5-1: SerialNumber: 5543733343735130C082
[ 1257.935430] cdc_acm 5-1:1.0: ttyACM0: USB ACM device
[ 2398.824233] usb 5-1: USB disconnect, device number 4
[ 2402.210473] usb 5-1: new full-speed USB device number 5 using ohci-pci
[ 2402.383729] usb 5-1: New USB device found, idVendor=23c1, idProduct=d314
[ 2402.383741] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[ 2402.383749] usb 5-1: Product: The Replicator
[ 2402.383755] usb 5-1: Manufacturer: MakerBot Industries
[ 2402.383760] usb 5-1: SerialNumber: 5543733343735130C082
[ 2402.386240] cdc_acm 5-1:1.0: ttyACM0: USB ACM device`

I've done:
sudo usermod -a -G tty $USER
sudo usermod -a -G dialout $USER
cd /dev/serial/by-id
sudo chown <me> usb*
sudo chgrp <me> usb*
sudo gpasswd -a <me> dialout
rebooted
but still it won't connect.

I've been trying and searching for 6 weeks now.

nallath, 2016-05-26:
It's not supporte by Cura, sorry. They use a different protocol.

closed 2016-05-26 by nallath

GettingNowhere, 2016-05-26:
What's that supposed to mean?  What's not supported?

Ghostkeeper, 2016-05-27:
Makerbot printers are running on firmware that's called Sailfish, which split off from Marlin a few years ago. The API of Sailfish is no longer compatible with that of Marlin. Cura only knows the API of Marlin, so it doesn't know how to talk to Sailfish. There is no easy way to support that since their firmware is closed source and the API unknown to us.

GettingNowhere, 2016-05-27:
Thank you for the explanation.
I have now removed Cura and won't bother you again.