daid/LegacyCura #1485: Cura 15.04.5 - Print speed.

E-492, 2016-05-30:
Hello.
I have this problem.
I set:
Print speed: 70 mm/s (in basic configuration)
Travel speed: 80 mm/s (in advanced configuration)
Bottom layer speed: 8 mm/s (in advenced configuration)
Infill speed: 45 mm/s (in advanced configuration)
Top/Bottom speed: 10 mm/s (in advenced configuration)
Outer shell speed: 10 mm/s (in advanced configuration)
Inner shell spped: 40 mm/s (in advanced configuration)
but only "Travel speed", "Bottom layer speed" and "Infill speed" values are used to generate gcode file.
The other "Top/Bottom spped", "Outer shell speed" and "Inner shell speed" are not used to generate gcode file.
Their values are always the same of "Bottom layer speed" o double of this value.
In this case the time to print the objects is too long because the speed is too slow.
I can't change manually every value in a very long gcode file.
Anyone can help me to resolve this problem?
Thanks for the attention.

Ghostkeeper, 2016-05-31:
Sounds like a bug to me, if that is the case. It should use the other three too.

rparrett, 2016-06-07:
I think that I'm seeing the same issue, but in Cura 2.1.1

In my case, I was slicing at 100% infill, and the issue seemed to be resolved by slicing at 99% infill instead.

It seems that 100% infill layers in the middle of the model are considered "top/bottom layers" when speed is calculated.

STL:
[14-Corner_Bracket.zip](https://github.com/daid/Cura/files/301668/14-Corner_Bracket.zip)

GCode @ 100% Infill:
[gk100.zip](https://github.com/daid/Cura/files/301669/gk100.zip)

GCode @ 99% Infill:
[gk99.zip](https://github.com/daid/Cura/files/301670/gk99.zip)

Speed Settings:
[image]

Gcode Viewer @ 100% infill:
[image]

Gcode Viewer @ 99% infill:
[image]

Full cura profile:
[infill-speed-profile.zip](https://github.com/daid/Cura/files/301672/infill-speed-profile.zip)

BagelOrb, 2016-06-07:
@Rob Parret, that is aan unrelated issue. Please file issues for Cura 2 in
the corresponding repository: github.com/Ultimaker/Cura

@OP please try Cura 2.1

op, 2016-06-07:
@BagelOrb Thanks. ;)