daid/LegacyCura #1486: Prints not starting correctly, printer head moves too low?

ThatWillisBillis, 2016-05-30:
I've been using Cura with my Printrbot Simple Metal for the good part of a year now, but I'm now having huge issues printing anything. I opened Cura today and was notified that there was an update available. I updated Cura and started to print something, but it didn't start the print correctly. As anyone with a Printrbot Simple is likely aware, there is an auto-leveling probe on the printing head, near the printer nozzle. Normally, the head homes itself in three of the four corners of the printer bed, then starts printing. 
    After updating Cura, however, the printer head only homes itself to the first of the three corners. It then raises up a few millimeters more that it normally would, and begins to print. Furthermore, the printer head drops too low and pushes onto the scotch tape I have covering the bed. If not for the tape, the printer nozzle would have gouged into my printer head.
    I've reinstalled Cura twice and reconfigured my printer inside Cura three times, but I can't make anything print correctly. Here's a link if you want to see the printer in action (you'll notice that the printer head doesn't drop down low, like I had mentioned. I have changed my z-offset to 0 so I don't damage anything. This made the nozzle seem to be the correct height, but as you can see in the video, the filament pushed down too far on the edge of the print closest to the camera): https://www.youtube.com/watch?v=6E5wr5--Nu8
    Any help is greatly appreciated, as this problem is a huge pain in the neck.

nallath, 2016-05-31:
Well, we don't officially support anything that's a non-ultimaker printer. All of those printers have been added by the comunity.
As far as I'm aware, cura does nothing with the bed leveling of any printer, so I'm at a loss as to why cura could make this behaviour change.

ThatWillisBillis, 2016-06-01:
It doesn't really seem like Cura did anything to the bed leveling itself, but instead blocked the printrbot firmware from running its homing commands. Did the gcode structure that Cura generates change at all with the last update?

daid, 2016-06-01:
Nothing changed in the 15.04.x releases except for profiles and firmware for the Ultimaker2+

The problem is more likely that the default profile for this machine is wrong and the user corrected for this in his previous install and forgot to correct for it now.

ThatWillisBillis, 2016-06-02:
Thanks for the help. I realize it could take a long time for Cura to be updated again, so does anyone have some suggestions in the meantime?

Ghostkeeper, 2016-06-05:
I'd suggest trying the 2.1 version (which is to be the future of Cura), or if all fails revert to an older version. Both can be downloaded from here: https://ultimaker.com/en/products/cura-software/list