daid/LegacyCura #1488: the bed leveling wizard

Gaming1Up, 2016-05-31:
when i use it for my printrbot it doesn't work.
i press the down buttons, it goes up. when i press the up buttons it does nothing. and i can figure it out, even when it starts the print cura just keeps moving the head up at each corner. it goes to a corner, goes up, and up again, next corner, same thing it does this for each corner then starts the print a few inches off the bed and i cant figure out why. repetier host does it just fine, but i like cura better.

daid, 2016-06-01:
The bed leveling wizard is designed for the Ultimaker. So "up" is moving the bed "up", as the Printrbot moves the head, it explains that inversion.

We don't own printrbot printers, so there is nothing we can do here to help. I suggest asking on the printrbot forums for assistance on this matter.

Brasshopper, 2016-08-11:
What would it take to make the bed leveling wizard a more general tool?  Would it have to be added as a separate menu item or could the current one be modified without offending the people who wrote it for Ultimaker?  I wrote a long issue before I read this answer.

If it really is only for the ultimaker (I noted the button inversion as well) and it only works incidentally on other printers, maybe it should say so on the first menu.  It actually says it will move the head to each corner.