daid/LegacyCura #1490: Infill Density = 100%: Bottom Layers goes berserk

demisjohn, 2016-06-03:
Cure 2.1.1

If I set `Infill Density = 100%`, `Bottom Layers` gets set to `999999`.  Any value less than 100% and the value is normal.

[image]
[image]

nallath, 2016-06-03:
This is by design. We fake 100 infill density by simply setting the number of bottom layers to a very high number. I do agree it's not the nicest solution, but if I recall correctly this made the engine faster.

closed 2016-06-03 by nallath

BagelOrb, 2016-06-03:
This also makes printing faster and more accurate, since no infill overlap
is applied.

daid, 2016-06-03:
Note, old Cura (15.04 and older) did the same, but without showing you.