daid/LegacyCura #1492: error while printing stops the process

jeetdeogun, 2016-06-05:
while printing from the software on my reprap 3d omni printer the printing stops during anytime of the process from 10-80 % and the error log has following errors
Communication timeout during printing, forcing a line
Send: M105 T0
Communication timeout during printing, forcing a line
Send: M105 T0
Communication timeout during printing, forcing a line
Send: M105 T0
Changing monitoring state from 'Printing' to 'Operational'
Send: M105 T0
Changing monitoring state from 'Operational' to 'Printing'
Send: N0M110*3
Send: N1M190 S70.000000*80
Send: N2M109 S210.000000*103
Send: N3G21*57
Communication timeout during printing, forcing a line
Send: M105 T0
Communication timeout during printing, forcing a line
Send: M105 T0
Communication timeout during printing, forcing a line
Send: M105 T0
Communication timeout during printing, forcing a line
Send: M105 T0
Communication timeout during printing, forcing a line
Send: M105 T0
Communication timeout during printing, forcing a line
Send: M105 T0
Communication timeout during printing, forcing a line
Send: M105 T0
Unexpected error while reading serial port: SerialException: 'call to ClearCommError failed' @ machineCom.py:_readline:516
Changing monitoring state from 'Printing' to 'Error: SerialException: 'call to ClearComm...'
Connection closed, closing down monitor

please help me solve the issue

SamPayet, 2017-07-27:
Hi, today i get the very same issue... Do you fix it on your printer?

daid, 2017-07-29:
The problem is with USB, not with the software.

closed 2017-07-29 by daid

DonYum, 2017-08-15:
@daid why?

daid, 2017-08-15:
Because "Unexpected error while reading serial port: SerialException: 'call to ClearCommError failed' @ machineCom.py:_readline:516"
Is straight from the USB serial driver, and I've tried everything I could to prevent this, and failed. I have a laptop that always has this error within 10 minutes.