daid/LegacyCura #1493: Can't setup 'Other' printer under Linux. Setup screen not large enough.

Lloyd-H1, 2016-06-06:
I went to setup a new 3D printer in Cura. It happens to be a Wanhoa i3. I get to the 1st list of printers in the Configuration Wizard and check 'Other' at the bottom. I get to the 2nd list of printers in the Configuration Wizard, and the bottom part of the list is not showing. The last printer that I can see is RigidBotBig. Everything under that is hidden. Nor is there any way to scroll down or expand the configuration window.

Running Cura 15.04.5 on Linux Mint 17.3

[cura2016june06](https://cloud.githubusercontent.com/assets/15940631/15815700/6ff5a7f0-2b94-11e6-8f62-4ffac8fda588.jpg)

Lloyd-H1, 2016-06-10:
I just installed Cura 15.04.6 on my Linux Mint 17.3 computer. Still have the same problem. Can't install new "Other" printer. It won't scroll at all to show bottom of menu.

daid, 2016-06-10:
As we're releasing Cura 2.1 right now, we're not planning on fixing the 15.04.x releases for this Linux issue.

closed 2016-06-10 by daid