daid/LegacyCura #1495: Adding Energy Estimate

fergussonb, 2016-06-07:
I am trying to add an estimate to how much energy will be needed to run a job. Can anyone point me to where they calculate the time and material estimates? Thanks in advance.

BagelOrb, 2016-06-07:
Time: https://github.com/Ultimaker/CuraEngine/blob/master/src/timeEstimate.cpp
Material is not so much an estimate; it's more like converting the difference in E-values, so I think that won't be that interesting to you.

In order to get an accurate energy estimate, I think you should calculate the energy for every move, then add up the energy to keep the hotend and bed heated based on the time estimates.

BagelOrb, 2016-06-07:
Note that TimeMaterialEstimates (https://github.com/Ultimaker/CuraEngine/blob/master/src/gcodePlanner.h#L56) are naive estimates, which are far from accurate.These are used in an earlier stage of processing where the eventual paths aren't fully known yet.

fergussonb, 2016-06-08:
Thanks do you know which part of the Cura GUI code displays the time estimate?

BagelOrb, 2016-06-08:
You should take a look at the material cost calculator plugin
https://ultimaker.com/en/community/16693-print-cost-calculator-plugin

closed 2016-06-08 by fergussonb