daid/LegacyCura #1497: Cura 15.4.6

Sangra69, 2016-06-09:
Ever since this crazy update this software has not be able to let me print. It keeps saving to an SD card file. I DON'T HAVE AN SD CARD SLOT ON MY PRINTER. I changed all the settings to turn off SD card support. All I want to do is access the UI printer interface to control the extruder. This software was fine before the new update. I am using a Robo3d R1.

Sangra69, 2016-06-09:
It keeps on wanting to open .ini files as well. It won't let me do anything else. This software is crap now it seems.

BagelOrb, 2016-06-09:
Maybe you should try out Cura 2.1
It's officially released now..

daid, 2016-06-10:
Replace your machine with an Ultimaker 2+?

Or, did you try rebooting windows? Windows is extremely bad with USB serial port handling. Which is why Cura (and Ultimaker) prefers SD printing over USB printing.

daid, 2016-06-10:
Note that there where limited changes in the 15.04.x releases, only small tweaks and profile updates for the Ultimaker2 line of printers. So I'm 100% convinced that it's not a change we made that is breaking your workflow.

closed 2016-06-10 by daid