daid/LegacyCura #1498: HOW COME THERE IS NEVER SUPPORT MATERIAL BETWEEN 2 PLASTIC LAYERS

tecnicolys, 2016-06-10:
Hello

Since I started using Cura every time I have a part that has some protruding tags, and these tags are positioned over some model geometry and not over the base plate, supporting material is never created. As you see the picture below, I would like to have some support material even on the area that is marked in green
.
[image]

Thank you

BagelOrb, 2016-06-10:
Try Cura 2.1! Its support generation mechanism has totally been revised. In
the new Curaq you will have way less problems such as the above.

daid, 2016-06-10:
Could also that you have "touching buildplate" support selected.

closed 2016-06-10 by daid