daid/LegacyCura #1500: Holes are missing

tius2000, 2016-06-13:
Holes are visibile in normal view and x-ray, but missing in layers and gcode

[Uploading holes-missing.zip…]()

Ghostkeeper, 2016-06-14:
Your model is broken. You can see in x-ray that there are extra walls inside the model. This could do strange things to a slicer. It assumes that the mesh separates the inside of the model from the outside, and thus that one of the sides of that extra wall has to be the outside of the model. And it all goes messy from there.

I suggest that you repair your model with a repair tool or using 3D modelling software.

tius2000, 2016-06-14:
Dear @Ghostkeeper, thank you for your fast reply and suggestions! 

I repaired the model in Sketchup and I think all extra walls are gone now, it looks fine to me in x-ray: 
https://www.dropbox.com/s/hwkyunyp0ozy5f2/holes-missing-v2.dae?dl=0

However, the path still seems to be wrong - is the model still broken?

BagelOrb, 2016-06-14:
Yes, some faces are inverted. No faces should look blue from the outside.

However, Cura can fix this: turn off "Union Overlapping Volumes"

Or just load the fixed stl: 
[Untitled-57607ebc2.zip](https://github.com/daid/Cura/files/315055/Untitled-57607ebc2.zip)
;)

BagelOrb, 2016-06-14:
Note that it's a bad idea to make the holes in the bottom so big that they coincide with the edges. The first layer will now remove the edge between the hole and the outside because it's too thin.

tius2000, 2016-06-15:
Dear @BagelOrb, thank you very much, it worked :+1:  

The option was named "Expert config > Combine everything (Type-A)" in my version of Cura (15.04.06)

closed 2016-06-15 by nallath