daid/LegacyCura #1504: can add first layer temperature setting in CURA Software ?

kokutoukiritsugu, 2016-06-16:
like slic3r 
[image]

some cheap filament In 215C first layer very bad.

old cura:
use Tweak At Z process so slow.
This plugins process whole gcode file.
old computer use this plugin need what a lot of time.
Goblin Racial: time is money firend !
[image]

new cura:
i only manual edit gcode:
[image]

BagelOrb, 2016-06-16:
I don't see any use in adding a first layer extruder temperature. That
doesn't make sense to me.

However, the first layer bed temperature does make sense to me.

We're definitely not going to fix this in Cura 15.x, but we might some time
in Cura 2.x.

Currently we're too busy with other features, but you're always welcome to
write a pull request,,

sp847242, 2016-11-28:
BagelOrb,
I wanted to weigh in on this and explain a bit about why I used a different first-layer temperature in a different software package. (I hope it's ok to comment on an old thread; better than starting a new one I guess.)

I did have a heated bed, but I also found that setting the first layer to a high temperature in order to reduce its viscosity allowed it to spread out a bit, and in turn it would adhere more strongly to the bed.

A higher first-layer bed temperature may do the job just as well though. The first printer I used did not have the ability to change the bed temperature, it was only on/off, and I never got around to installing my own temperature controller.
I have a DeltaWasp printer now, and it does have bed temperature control.

BagelOrb, 2016-11-28:
Implemented in Cura 2.4

closed 2016-11-28 by BagelOrb