daid/LegacyCura #1508: Printrbot on cura not moving when the print is canceled.

Thorgodof4, 2016-06-28:
So i just got my first printer, printrbot play, today and i need some help. When I cancel a print the extruder dose not move. this the code:
;End GCode
M104 S0                     ;extruder heater off
M140 S0                     ;heated bed heater off (if you have it)

G91                                    ;relative positioning
G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F{travel_speed} ;move Z up a bit and retract filament even more
G28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way

M84                         ;steppers off
G90                         ;absolute positioning
;{profile_string}

: So if you can help please do so thanks.

Ghostkeeper, 2016-06-28:
I think this is an issue with Printrbot, not with Cura. The g-code to execute when you cancel your print is not in the file, unless it's actually skipping to `;End GCode` but then it should properly retract the filament and move to 0 and stuff. In either case, it's not in Cura's power to fix.

Thorgodof4, 2016-06-28:
I do not understand what you mean.

BagelOrb, 2016-06-28:
The end gcode is only executed when you actually reach the end of your
print - not when you cancel a print.

Cura has no functionality to execute gcode the moment you cancel a print.
At Ultimaker we just let the printer decide what to do when a print job is
cancelled, so the slicer is not involved when cancelling a print - the
printer is responsible.

<https://github.com/daid/Cura/issues/1508#issuecomment-229106593>, or mute
the thread