daid/LegacyCura #1509: Cannot package cura for windows

Cyndent, 2016-06-28:
I run package.sh on gitbrash.
It runs well untill build cura engine .
it gives the info:
`Extracting ejectmedia.zip Win32
make: Entering directory `D:/cura-development/Cura-SteamEngine/CuraEngine'
make: Leaving directory `D:/cura-development/Cura-SteamEngine/CuraEngine'
make: *** No targets specified and no makefile found.  Stop.
`
I check the resouce on Form and the issues,but cannot find how to resolve it .I do not know what to do.Can somebody help me?

Ghostkeeper, 2016-07-04:
Did you manage to find an answer?

I'm sorry we couldn't help much. The error message says pretty much nothing.

Cyndent, 2016-07-05:
>

Cyndent, 2016-07-05:
Still do not know what goes wrong, but I packaged it.
Maybe something wrong with Engine building. When I run "package.sh",it stop at MAKE engine. 
But I comment out this code ,move engine.exe and it stop at some strange place. 
I package it with NSIS depend on "package.sh" and installer.nsi. And it worked.
Now I'm confused about the envrionement. Did I need Microsoft VC++ to build the engine? It's used in new version of Cura. And I didnot install it on my laptop.

Ghostkeeper, 2016-07-05:
You shouldn't need Microsoft VC++. It's made to be compiled with GCC/MinGW and uses no special MS-things. It's just some libraries in the front-end that need it.