daid/LegacyCura #1511: How can i connect my 3D printer?

lin6328, 2016-06-29:
I am building a 3D printer.And my 3D printer based on STM32. I do not know what i should write that i can connect the cura .Cura and 3D printer needs what information,

Ghostkeeper, 2016-06-29:
This is what I understood...

Cura connects via a COM port. It tries both baud rates 115200 and 250000. To verify a connection, Cura sends a temperature request to the printer: "M105". It waits for an answer. It checks whether the answer is a number.

To print, Cura sends the g-code over the COM port.

This all works well if the firmware is any Marlin variant.

lin6328, 2016-06-29:
Thank you .I try it now.

lin6328, 2016-06-29:
I can connect 3D printer now. But when i kick printing,the machine doesn't work.However when i send G-code file through serial debugging tools,it works.I think cura maybe have send other message to 3D printer,and 3D printer will reply message to cura.Now,the question is what message they send··· @Ghostkeeper thank you very much.

lin6328, 2016-06-30:
My 3D printer is working now,it can control by cura.But it will work wrong sometimes.I do not check the code line.Is the problem in here?

BagelOrb, 2016-06-30:
Depends on what the problem is.
What problem are you experiencing?

lin6328, 2016-06-30:
Marlin use 'N' to counter the gcode line.If the gcode is wrong,marlin will reply the wrong line to cura.But I do not use 'N' to counter.I think cura send the gcode one by one , it should not to be wrong.So whar is problem? @BagelOrb

BagelOrb, 2016-07-04:
I cannot say where the problem is if I don't know what the problem is.
What problem are you experiencing?