daid/LegacyCura #1512: gcode file different from the stl file (cura 15.04.6)

softysoft, 2016-06-30:
Hi,

At the end of printing with my Ultimaker 2, I got a box full with PLA: the gcode file does not match the stl file.

Please see attached zip file (the zip file include stl and gcode files)

I don't understand why.

Thanks for answer.

[box.zip](https://github.com/daid/Cura/files/341967/box.zip)

Ghostkeeper, 2016-06-30:
Some faces of your mesh are inside-out. This causes Cura to be confused what the inside of your model is and what the outside is. I've fixed the model for you in Blender.

[box-fixed.zip](https://github.com/daid/Cura/files/342061/box-fixed.zip)

softysoft, 2016-07-04:
Thanks for answer,

I don't understand how I create the problem: in sketchup, I made an empty box (like a soap box), then I drill holes, then I cut the box (the role of the box is to contain a PCB). One piece was OK in Cura, and the other one, no, although the two sides appear correct in the stl files.

?

B.Aouizerat
SOFTYSOFT
Tél.:   +33 4 72 65 00 00
?Gsm:  +33 6 60 38 27 50
Mail:  b.aouizerat@softysoft.com

daid, 2016-07-04:
The problem is sketchup, it's very bad for 3D modeling physical models. Because it does not keep a proper representation of what is "inside" and what is "outside". Causing Cura to be confused.