daid/LegacyCura #1513: some item placement does not work (cura-git-2.1.0.1366.gdc2384c)

hmalzahn, 2016-06-30:
I'm using the current git branch under Arch Linux. Recently I chanced into a problem with item placement that wasn't there a few builds before. Some things "wander" on the buildplate when duplicated. It's easy to reproduce:

1. clean buildplate
2. load an item like this one
[cura-testcube.zip](https://github.com/daid/Cura/files/342056/cura-testcube.zip)
3. right click on it and select "duplicate"

after which it "wanders off" the buildplate

BagelOrb, 2016-06-30:
Try the new Cura: cura 2.
We don't develop on cura 15.x anymore

hmalzahn, 2016-07-01:
Like it says in the subject line and in my text: I am using the **current git branch of Cura 2.1**, compiled from the master branch of git://github.com/Ultimaker/Cura.git. Who said anything about Cura 15?

The described effect started to happen after one of the latest changes, 2.1.0.1354 I think, earlier git builds work just fine.

Ghostkeeper, 2016-07-01:
You're posting in the old repository of Daid, which was discontinued. Bugs on Cura 2 and further should be reported at http://github.com/Ultimaker/Cura .

That said, we know about this bug in Cura 2.1. It occurs in edge cases when edges of the collision areas line up. This was the case when duplicating, and when the mesh was re-scaled to fit the volume. This is fixed in version 2.2 and currently waiting for QA to verify it as fixed.

> I am using the current git branch of Cura 2.1, compiled from the master branch of git://github.com/Ultimaker/Cura.git

I'm a bit confused of how you say that you use the current git branch of 2.1 but compiled from the master branch. You compiled from the master branch and then use the source code from 2.1?

hmalzahn, 2016-07-01:
Oops, I guess I used the wrong bookmark, hence the wrong repository. Seeing many issues regarding version 15 and none about 2.1 might have given it away... anyway, thanks for confirming that it's a known issue, so I'll just close this one.

As to "2.1" and "master", I simply worded that confusingly. I am building from "master", but the version numbers "git describe" outputs are "2.1.0.nnnn", so that's what I meant.

closed 2016-07-01 by hmalzahn