daid/LegacyCura #1514: why it print the layer at the wrong place

snuchitp, 2016-07-01:
I need to use this version because of my 3D printer. New version not support DeltaBot.
Please do not close my issue. I need some help

I used cura v 15.04.6 to display my model.  From layer 1-9, it looked correct that built each layer up on top. But from layer 10- the end  started to build from the bottom of the model. The detailed lines are different too.

layer 0-9
[image]

layer 10 -14
[image]

Thanks

BagelOrb, 2016-07-01:
It's not totally clear, but I think you are talking about the fact that the
lower layers are displayed as thin lines.

Note that this is only a display issue. Lower layers show thin lines so
that the rendering can go faster. This has nothing to do with the actual
gcode.

Only the top few layers show how the gcode actually is.

<https://cloud.githubusercontent.com/assets/12837769/16528681/57645856-3f8c-11e6-9dc0-2d558b7e7f1e.png>

<https://cloud.githubusercontent.com/assets/12837769/16528683/5bf05032-3f8c-11e6-9692-c90ec3e63927.png>

snuchitp, 2016-07-01:
layer 0-2 should be think lines
layer 3-21 should be thin lines
layer 22-24 should be thick lines 

The model should look like this

[image]

snuchitp, 2016-07-01:
layer 0-2 should be thick lines

Top view

[image]

side view

[image]

snuchitp, 2016-07-01:
layer 3-21 should be thick lines

Top view

[image]

side view

[image]

BagelOrb, 2016-07-01:
I don't really understand. Could you attach your model in a zip file?

Ghostkeeper, 2016-07-03:
I suspect that what happens is that the new layers actually do get built on top of the old ones, but since the camera moves along with the highest layer, it looks like there are new layers on the bottom when you scroll up.

daid, 2016-07-04:
This is a rendering optimalization. Lower layers then you are viewing are rendered quicker.

closed 2016-07-04 by daid