daid/LegacyCura #1516: QOpenGLShader::compile(Fragment): 0:7(1): error: No precision specified in this scope for type `float'

cor-mt, 2016-07-11:
I created ebuilds for installing cura 2.1.0 and all it's libraries on gentoo/funtoo https://github.com/cor-mt/portage-overlay , it install fine but when running there is no OpenGL stuf.

```
/usr/bin/cura
ATTENTION: default value of option vblank_mode overridden by environment.
2016-07-11 07:25:57,350 - INFO - Loaded plugin ConsoleLogger
2016-07-11 07:25:57,380 - INFO - Loaded plugin CuraEngineBackend
2016-07-11 07:25:57,384 - INFO - Loaded plugin WireframeView
2016-07-11 07:25:57,388 - INFO - Loaded plugin SimpleView
2016-07-11 07:25:57,440 - INFO - Checking for new version of cura
2016-07-11 07:25:57,446 - INFO - Loaded plugin UpdateChecker
2016-07-11 07:25:57,468 - INFO - Loaded plugin TranslateTool
2016-07-11 07:25:57,474 - INFO - Loaded plugin SelectionTool
2016-07-11 07:25:57,490 - INFO - Loaded plugin ScaleTool
2016-07-11 07:25:57,506 - INFO - Loaded plugin RotateTool
2016-07-11 07:25:57,515 - INFO - Loaded plugin MirrorTool
2016-07-11 07:25:57,525 - INFO - Loaded plugin CameraTool
2016-07-11 07:25:57,531 - INFO - Loaded plugin FileLogger
2016-07-11 07:25:57,586 - INFO - Loaded plugin STLWriter
2016-07-11 07:25:57,591 - INFO - Loaded plugin STLReader
2016-07-11 07:25:57,599 - INFO - Loaded plugin OBJWriter
2016-07-11 07:25:57,607 - INFO - Loaded plugin OBJReader
2016-07-11 07:25:57,628 - INFO - Loaded plugin 3MFWriter
2016-07-11 07:25:57,629 - WARNING - Plugin ConsoleLogger was already loaded
2016-07-11 07:25:57,641 - INFO - Loaded plugin LocalFileOutputDevice
2016-07-11 07:25:57,663 - ERROR - Import error loading module USBPrinting: No module named 'serial'
2016-07-11 07:25:57,663 - ERROR - Could not find plugin USBPrinting
2016-07-11 07:25:57,677 - ERROR - Import error loading module USBPrinting: No module named 'serial'
2016-07-11 07:25:57,689 - INFO - Loaded plugin LayerView
2016-07-11 07:25:57,696 - INFO - Loaded plugin GCodeWriter
2016-07-11 07:25:57,696 - WARNING - Plugin CuraEngineBackend was already loaded
2016-07-11 07:25:57,704 - INFO - Loaded plugin XRayView
2016-07-11 07:25:57,711 - INFO - Loaded plugin SolidView
2016-07-11 07:25:57,719 - INFO - Loaded plugin SliceInfoPlugin
2016-07-11 07:25:57,732 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-07-11 07:25:57,747 - INFO - Loaded plugin PerObjectSettingsTool
2016-07-11 07:25:57,755 - INFO - Loaded plugin LegacyProfileReader
2016-07-11 07:25:57,769 - INFO - Loaded plugin ImageReader
2016-07-11 07:25:57,776 - INFO - Loaded plugin GCodeProfileReader
2016-07-11 07:25:57,784 - INFO - Loaded plugin CuraProfileWriter
2016-07-11 07:25:57,792 - INFO - Loaded plugin CuraProfileReader
2016-07-11 07:25:57,801 - INFO - Loaded plugin ChangeLogPlugin
2016-07-11 07:25:57,809 - INFO - Loaded plugin AutoSave
2016-07-11 07:25:57,818 - INFO - Loaded plugin 3MFReader
2016-07-11 07:25:57,819 - INFO - Command line arguments: {'file': [], 'disable-textures': False, 'debug-mode': False, 'external-backend': False}
2016-07-11 07:25:58,071 - WARNING - No active machine found when trying to save setting visibility
2016-07-11 07:25:58,073 - DEBUG - Attempting to kill the engine process
2016-07-11 07:25:58,229 - INFO - Started engine process: /usr/bin/CuraEngine
2016-07-11 07:25:58,242 - DEBUG - Backend connected on port 49674
2016-07-11 07:25:58,243 - DEBUG - Setting active view to SolidView
2016-07-11 07:25:58,531 - DEBUG - Calculating normals took 0.0015392303466796875 seconds
2016-07-11 07:25:58,531 - DEBUG - Loaded a mesh with 8052 vertices
2016-07-11 07:25:59,203 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
2016-07-11 07:26:00,308 - DEBUG - Initialized OpenGL subsystems.
2016-07-11 07:26:00,308 - DEBUG - OpenGL Version:  OpenGL ES 3.0 Mesa 11.2.2
2016-07-11 07:26:00,308 - DEBUG - OpenGL Vendor:   X.Org
2016-07-11 07:26:00,308 - DEBUG - OpenGL Renderer: Gallium 0.4 on AMD TAHITI (DRM 2.43.0, LLVM 3.8.0)
QOpenGLShader::compile(Fragment): 0:7(1): error: No precision specified in this scope for type `float'
0:8(1): error: No precision specified in this scope for type `float'
0:12(1): error: No precision specified in this scope for type `float'
0:13(1): error: No precision specified in this scope for type `vec4'
0:15(1): error: No precision specified in this scope for type `vec2'
0:19(1): error: No precision specified in this scope for type `vec3'
0:20(1): error: No precision specified in this scope for type `vec3'
0:21(1): error: No precision specified in this scope for type `vec3'
0:29(1): error: No precision specified in this scope for type `vec4'
0:30(1): error: No precision specified in this scope for type `vec4'
0:34(1): error: No precision specified in this scope for type `vec4'
0:37(1): error: No precision specified in this scope for type `vec4'
0:41(1): error: No precision specified in this scope for type `vec4'

*** Problematic Fragment shader source code ***
#ifndef GL_FRAGMENT_PRECISION_HIGH
#define highp mediump
#endif
#line 1

uniform sampler2D u_layer0;
uniform sampler2D u_layer1;
uniform sampler2D u_layer2;
uniform sampler2D u_layer3;

uniform float u_imageWidth;
uniform float u_imageHeight;

uniform vec2 u_offset[9];

uniform float u_outline_strength;
uniform vec4 u_outline_color;

varying vec2 v_uvs;

float kernel[9];

const vec3 x_axis = vec3(1.0, 0.0, 0.0);
const vec3 y_axis = vec3(0.0, 1.0, 0.0);
const vec3 z_axis = vec3(0.0, 0.0, 1.0);

void main()
{
kernel[0] = 0.0; kernel[1] = 1.0; kernel[2] = 0.0;
kernel[3] = 1.0; kernel[4] = -4.0; kernel[5] = 1.0;
kernel[6] = 0.0; kernel[7] = 1.0; kernel[8] = 0.0;

vec4 result = vec4(0.965, 0.965, 0.965, 1.0);
vec4 layer0 = texture2D(u_layer0, v_uvs);

result = layer0 * layer0.a + result * (1.0 - layer0.a);

vec4 sum = vec4(0.0);
for (int i = 0; i < 9; i++)
{
vec4 color = vec4(texture2D(u_layer1, v_uvs.xy + u_offset[i]).a);
sum += color * (kernel[i] / u_outline_strength);
}

vec4 layer1 = texture2D(u_layer1, v_uvs);
if((layer1.rgb == x_axis || layer1.rgb == y_axis || layer1.rgb == z_axis))
{
gl_FragColor = result;
}
else
{
gl_FragColor = mix(result, u_outline_color, abs(sum.a));
}
}
***
QOpenGLShader::link: "error: program lacks a fragment shader\n"
2016-07-11 07:26:00,336 - ERROR - Shader failed to compile!
QOpenGLShaderProgram::uniformLocation( u_layer0 ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_layer1 ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_layer2 ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_layer3 ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_outline_strength ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_outline_color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_offset ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_modelViewProjectionMatrix ): shader program is not linked
QOpenGLShaderProgram::attributeLocation( a_vertex ): shader program is not linked
QOpenGLShaderProgram::attributeLocation( a_uvs ): shader program is not linked
2016-07-11 07:26:11,462 - DEBUG - Autosaving preferences, instances and profiles
2016-07-11 07:26:23,888 - DEBUG - Shutting down cura
```

daid, 2016-07-11:
Cura 15.04 does not support hardware that only supports "OpenGL ES"

You'll have to switch to the newer Cura 2.0, which people did get running on OpenGL ES.

closed 2016-07-11 by daid

BagelOrb, 2016-07-11:
You should log this under Ultimaker/Cura rather than daid/cura

kg1011, 2018-12-29:
You can try to add "precision mediump float" before 1st line, and remove all float in your src.