daid/LegacyCura #1521: Packaging Cura 15.04.6 for MacOS

remelbourne, 2016-07-19:
I am trying to package Cura for my Mac which is running version 10.6.8. I have followed the required steps on the readme page and everything has worked fine up until I run the command "./package.sh"

I keep getting the following error message:

[image]

I searched the problem online and some said the problem would be fixed by changing -std=c++11 with -std=c++0x but that did not fix the problem.

My g++ version is:

[image]

I am unable to update to a newer g++ version and I was wondering if that is the problem.

Any help would be much appreciated!

Ghostkeeper, 2016-07-20:
I don't think GCC v4.2.1 has any C++11 support. It was released in July 2007, and the 11 stands for 2011.

Perhaps if you are unable to update to newer g++ versions, you could consider using Clang?

daid, 2016-07-20:
Building for OSX is just a world of pain. You can thank Apple for that.

Note that we are no longer planning to update the 15.06 version of Cura, as we are switching to Cura 2 for all development.

closed 2016-07-20 by daid