daid/LegacyCura #1522: When I change the printing speed in Cura the estimated time to finish stay's the same.

heideict, 2016-07-24:
Hello,

I have an Ultimaker 2+
When I change layer settings and such. there is no problem Cura wil adjust the estimated time correctly.
But when I change the printing speed from 30 mm/s to for example 50 mm/s the time stays exactly the same als for 30 mm/s.

Ghostkeeper, 2016-07-25:
Maybe you're hitting the minimum layer time. Is it a small print?
By default it can't take less than 5 seconds per layer.

heideict, 2016-07-26:
Thats not it.
if i print big items the problem stays the same.
I din't change anything in de software.
Just like it was after download.

Ghostkeeper, 2016-07-27:
I can't reproduce the problem with any normal-sized mesh.

The only other thing I can think of is that it might hit acceleration maxima, i.e. it is accelerating or decelerating the whole time and isn't able to reach the target speed, but for 30mm/s that would take at most 0.016̅s so I wouldn't think that's the problem.

Are you sure that the g-code you are getting is actually different? Is it reslicing properly?

BagelOrb, 2016-07-27:
Which version of Cura are you using?

heideict, 2016-07-27:
I am using Cura 15.04.5

I have checked the Gcode, I looks like the Gcode isn't changed when I change the speed settings in Cura.

[print 50mm-s.txt](https://github.com/daid/Cura/files/385541/print.50mm-s.txt)
[print 30mm-s.txt](https://github.com/daid/Cura/files/385542/print.30mm-s.txt)

BagelOrb, 2016-07-27:
Cura 2.1 is out!

We're not developing on the old Cura any more.

heideict, 2016-07-27:
OK, Downloaded and installed the new version, 
problem is fixed. 
thank you for the support.

closed 2016-07-27 by BagelOrb