daid/LegacyCura #1523: machine_center_is_zero Causes Freeze

dr2391, 2016-07-27:
Just created a JSON file for my new Delta printer but every time I tried to slice a couple different objects that usually take 2-4 hours to print, Cura freezes up about half way through and I have to force quit. Smaller 4 min prints sliced fine.

Through process of elimination, I found that by removing or setting machine_center_is_zero to false, Cura slices the files with no problem. 

Can anyone reproduce this?

Ghostkeeper, 2016-07-27:
Tried this with 3D Benchy using a modified definition of the Ultimaker 2+, but was unable to reproduce it :(
I tried slicing Benchy in every of the four corners, as well as the centre and maximised.

BagelOrb, 2016-07-27:
Which version of Cura are you using?

Ghostkeeper, 2016-07-27:
See, the problem with replying to e-mails rather than on Github itself is that you might be replying to messages which have since been deleted or edited :yum:

BagelOrb, 2016-07-27:
I don't see how that's a problem here.

I think you are using Cura 2.x, so you should be reporting problems in the corresponding repo: https://github.com/Ultimaker/Cura/issues

closed 2016-07-27 by BagelOrb