daid/LegacyCura #1524: Make an extra border in scene view

aquibjaved, 2016-07-28:
I am changing the GUI part of Cura, I want to make an extra border on the sides of the build volume and another one is I want put some logo instead of " Basic" Button. How can I do it?

nallath, 2016-07-28:
There is no real easy answer to give here. I'd recommend taking a look at the source code and play around with it a bit.

Note that this cura is the old (and no longer supported version). These kind of GUI changes are easier to do in the new Cura.

closed 2016-07-28 by daid

aquibjaved, 2016-07-28:
can you Please give me the link from where I can download and build the new cura. And also please let me know how can I change the GUI of new one.

BagelOrb, 2016-07-28:
software.ultimaker.com

daid, 2016-07-28:
There is no step-to-step guide for this. If you cannot figure it out on your own, you are most likely also not capable of doing anything else without our help. Which means we are doing your work for you. And I don't think we get payed for that.

aquibjaved, 2016-07-29:
hmm, Alright thank you

Ghostkeeper, 2016-07-29:
Take a look at https://github.com/Ultimaker/cura-build . It provides CMake scripts that should build Cura 2 for you from the ground up. There's often a few extra steps to take depending on your operating system and the libraries Cura depends on, but with a bit of trickery anyone should be able to get it working eventually.

aquibjaved, 2016-07-29:
I am getting this error while building it. When I hit the command sudo make

CMakeFiles/Qt.dir/build.make:89: recipe for target 'Qt-prefix/src/Qt-stamp/Qt-download' failed
make[2]: *** [Qt-prefix/src/Qt-stamp/Qt-download] Error 1
CMakeFiles/Makefile2:736: recipe for target 'CMakeFiles/Qt.dir/all' failed
make[1]: *** [CMakeFiles/Qt.dir/all] Error 2
Makefile:105: recipe for target 'all' failed
make: *** [all] Error 2

Ghostkeeper, 2016-07-30:
That's a target to download one of the dependencies, the Qt GUI framework. Your snippet doesn't mention why the Qt-download target failed, but it might just be a lack of internet connection or a hiccup on Riverbank's servers.

At this point it's probably better to move any further problems with building Cura to the cura-build repository, as this all has little to do with Daid's Cura: https://github.com/Ultimaker/cura-build/issues