daid/LegacyCura #1525: Some elements in stl file missing in gcode

SpikeBagshaw, 2016-07-29:
Hi I have an stl file created in TCad 2015 which loses some of the ribs when converted to gcode in CURA.  Does anyone know why?
Thanks

[Marsden E Bridge.zip](https://github.com/daid/Cura/files/391013/Marsden.E.Bridge.zip)

BagelOrb, 2016-07-29:
Have a look at the x-ray view. Red parts mean that your model has problems:
holes or internal geometry.

Try fooling around with the mesh fixes in expert settings.

SpikeBagshaw, 2016-07-30:
Thanks.  I've used the x-ray view and it doesn't show any red unless I move the viewpoint inside the piece, then it's all red.  When I switch on the Black Magic->Follow mesh surface then it produces all the ribs in gcode but doesn't do any infill.  I'll try other combinations.

Ghostkeeper, 2016-07-30:
X-ray view works by colouring a pixel red when there is an odd number of surfaces beneath it, so if the model is correct then everything should appear red when your camera is inside the model :smiley:

I measured the width of those ribs and it appears that they are exactly 0.4mm wide. At a certain point, Cura disregards such details because it can't print anything that thin. Reducing the nozzle size helps (0.26 or lower made the ribs appear for me) but otherwise Cura is thinking there is no way for it to print those ribs.

It's worth mentioning that Cura 2 handles this case better. It still disregards any details that are smaller than the nozzle size, but doesn't need to fit two lines inside a rib, just one. Hence, a line width of 0.399mm makes all ribs appear.

SpikeBagshaw, 2016-07-31:
Thanks re X-ray view.  I've just built an i3 Prusa and am experimenting with what it will do.  I set the rib width to 0.4mm thinking it would print with 1 pass.  It's odd that it has included 3 ribs but not the rest.
Presumably setting the rib width to say 0.5 will allow 2 passes.  I'll try this.
I'll also try Cura 2 it looks to have a lot of interesting settings.  I'm looking for one that will give a better finish to flat areas, less regular that the diagonal pattern on my model. 
Many thanks.

Ghostkeeper, 2016-08-01:
You could try fiddling with the top/bottom line width in Cura 2. If you have a smaller line width than your nozzle size, the nozzle will hit part of the neighbouring skin line which might melt the lines together a bit.

There are different patterns for the skin in Cura 2: lines (traditional recipe), zig-zag and concentric. But I fear that none of these would really give a better surface finish than lines.