daid/LegacyCura #1526: Problem with moving objects on platform

bobharrison16, 2016-08-01:
I have been working fine until lately.  

When i load an object up and place it on the platform thats ok but then when you put another object on the same platform you cannot move it into position.

Can anyone advise?

Thanks

Ghostkeeper, 2016-08-01:
Dragging the object has no effect? Or is it the g-code that doesn't change?

bobharrison16, 2016-08-01:
Hi, dragging the second object is the problem.

Thanks Ghostkeeper.

BagelOrb, 2016-08-01:
Try out Cura 2!
Download it at software.ultimaker.com

Ghostkeeper, 2016-08-01:
Can't reproduce it here, in any case.

bobharrison16, 2016-08-01:
Sorry i dont understand

BagelOrb, 2016-08-01:
There is a newer version of Cura released; Cura 2.1 is newer than Cura 15.

If you have issues with that version of Cura, please log them at
https://github.com/Ultimaker/Cura/issues

Ghostkeeper, 2016-08-02:
This is what I mean. It works for me: http://dulek.net/work/DuplicateAndMove.ogv