daid/LegacyCura #1527: latest build, when installed changed default filament dia at 2.83 (or so) rather than 1.75

Paul-Bostwck, 2016-08-04:
I am not sure where such settings live. But it took some sniffing to finally figure it out.

BagelOrb, 2016-08-04:
Try Cura 2.1! You can download it at software.ultimaker.com

Paul-Bostwck, 2016-08-04:
Is that the next step above 15.04.6?

BagelOrb, 2016-08-04:
Yes.
Its a whole new Cura! The whole UI is new. It takes some getting used to,
but there are a lot more settings to tweak and a lot of new features.

Paul-Bostwck, 2016-08-04:
Wait from the link it looks like 2.1 IS 15.04.6…. Confusing

Paul-Bostwck, 2016-08-04:
Ok I loaded the new version and it does not have a Printrbot option… are they out of favor?

BagelOrb, 2016-08-04:
They just haven't been added to the new Cura yet. I don't know if printrbot
is already in the repo for the next release though. @ghostkeeper?

3rd party printers are generally included by community members who write
the appropriate json files.

Next release will be up in a month or so. Maybe earlier, maybe later.

>
> Try Cura 2.1! You can download it at software.ultimaker.com
>
>
> > I am not sure where such settings live. But it took some sniffing to
> > finally figure it out.
AIe9Ecv6Dlkf4muWuJ8W5R369TWqkW-wks5qck6QgaJpZM4JdGhS>
https://github.com/daid/Cura/issues/1527#issuecomment-237677587>, or mute
the thread <https://github.com/
auth/AQEr7cxQ62vzgZUW6W6AmKlq-qH9_h7dks5qck7-gaJpZM4JdGhS>.

<https://github.com/daid/Cura/issues/1527#issuecomment-237690635>, or mute
the thread

Paul-Bostwck, 2016-08-04:
OK… I’ll stick with the last build for now and wait for the new goodies down the line.

Ghostkeeper, 2016-08-05:
To actually answer your question though, the filament diameter for Printrbot shows up as 1.75 by default for me.
It's 2.85 for Ultimaker printers though. Perhaps you had the wrong printer selected?

Paul-Bostwck, 2016-08-05:
maybe… I ‘ll take a look. It did not step me through the setup wizard ((which I regularly faced in the earlier version so thanks for fixing that)

play by play:
Mac El Capitain OS
Quit Cura (15.04.6)
Find the package in downloads folder
Unzip
Open .dmg
Move Cura logo to alias of Appliations Folder
 (yes to “copy over other copy”)
Double Click on Cura icon in Applications Folder
Starts without setup wizard (thanks again)
Shows Filament size of 1.75 (WHOOT!)
Check Machine Menu “Printrbot Simple Metal”

I must have fat fingered the settings!

Ghostkeeper, 2016-08-05:
We have a Printrbot Simple definition in 2.2. Not in 2.1 yet. The Printrbot Simple definition has been tested to also work fine with the Simple Metal.