daid/LegacyCura #1528: Printer leveling wizard assumes a 3 screw bed

Brasshopper, 2016-08-11:
I have one of the Monoprice compact IIIP printers, and they supply a copy of Cura as the recommended slicer.  My bed was badly out of adjustment.  The first sample printed OK, but now I was, using different tape for the bed than supplied (it tore and I bought blue masking tape) and also it was falling farther out of calibration with each print.   The bed was too high, and every print stuck badly to the tape and the tape had to be replaced.  Additionally the first ot second layer simply didn't print.  Bad news.  So I decided to use the printer leveling wizard.

This printer has an adjustment screw in each corner, that is, it has 4 bed leveling adjustment screws..  The wizard assumes a three screw bed. It did not work to do a single adjustment on the back,  I tried to match turn for turn but in the end, the left side is probably 0.03 mm lower than the right, based on how the first layer of a raft print works. 

Would it be possible to make checking at each corner an option?  Or different patterns, like front center, (middle x, minimum Y), left (minimum X, middle y), right (maximum X, middle y) and back center (middle x, maximum y).   Again, this printer has corner screws that are blocked when the print head is in the corner, so for the front left ans right corners I had to mover the bed and then try and move it back to the corner.  The process didn't feel automatic.

If I could select from the current 3 hole pattern, a 4 corner pattern, a 4 center side pattern, my printer bed leveling life would be complete.  And I could use the 4 center for most adjusting, check it with the 4 corner, and get a really superior job.

nallath, 2016-08-11:
It would be possible, but it's up to the people who support machines with this kind of setup. We (read the people from Ultimaker) won't actively develop / support other machine setups, because we don't have time to do so (nor do we feel responsible for tweaking the user experience for competitors printers).
If someone builds it, we're more than happy to include it in the flow. In Cura 2.2 this is possible by adding MachineActions.

closed 2016-08-11 by nallath

Ghostkeeper, 2016-08-12:
Wonder how this would work with four screws, since when you adjust one corner downwards, the diagonally opposite corner would go up.

daid, 2016-08-12:
If you turn off the heated bed option, you will get 4 point leveling. Due to the UMO having 4 point leveling and the UMO+ having 3 point leveling. (Yes this is designed for just the UMO line of printers in mind)