daid/LegacyCura #1529: Config Wizard issue

MarkBruckard, 2016-08-12:
Hi.  I'm new to Cura and am running 15.04 on Linux.  I am trying to set up my Wombot 3D printer but I have run into an odd problem: The option I need is below the edge of the configuration wizard window.  I can't tab through the options, I can't resize the window, nothing.  
The Wombot instructions tell me to select "Other" at the bottom of this page, but as you can see, it is outside the limits of the window.
[image]

MarkBruckard, 2016-08-12:
Here is what I am suppose to see.
[image]

MarkBruckard, 2016-08-12:
Sorry, that should have been "Custom" not "Other" in my original post.

Ghostkeeper, 2016-08-12:
Duplicate of https://github.com/daid/Cura/issues/1418

daid, 2016-08-12:
As we are not planning to update the 15.04 anymore in preference for the new Cura 2.1, you'll have to use a workaround.

You can delete a bunch of machine profiles in /usr/share/cura/resources/machines to reduce this list and make the other option visible.

closed 2016-08-12 by daid

MarkBruckard, 2016-08-12:
Thanks.  All good now.  I just put the .ini file for my printer in that folder and moved the rest into a spares folder.  works now.  thanks.