daid/LegacyCura #1532: Crashing on initial situation of a large part on Ubuntu 14.04LTS (2.1.3)

opensourcefan, 2016-08-20:
I have a largish part that I need to rotate a couple times after loading it. If I can't get it right within the first few seconds Cura will crash closed. Right now it takes me about five times to get a large part situated due to me having to practice all the movements so I can do them quickly. It always crashes as it'a slicing. Seems to me that it's getting overwhelmed by all the movements and continued slicing and re-slicing. 

Can you point me to any log information that I can get for you that will help with this. My PC isn't a slouch and can handle almost anything I throw at it.

opensourcefan, 2016-08-20:
A couple other things I've noticed. Some of the time if I move a part off the build area it will crash immediately. 

Also if for some reason I'm able to do a bunch of movements without it crashing it seems to stop the slicing all together. It just sits at "Preparing to slice" forever.

opensourcefan, 2016-08-20:
cura
2016-08-19 18:40:16,007 - INFO - Loaded plugin ConsoleLogger
2016-08-19 18:40:16,026 - INFO - Loaded plugin CuraEngineBackend
2016-08-19 18:40:16,050 - INFO - Checking for new version of cura
2016-08-19 18:40:16,052 - INFO - Loaded plugin UpdateChecker
2016-08-19 18:40:16,067 - INFO - Loaded plugin RotateTool
2016-08-19 18:40:16,072 - INFO - Loaded plugin SelectionTool
2016-08-19 18:40:16,077 - INFO - Loaded plugin TranslateTool
2016-08-19 18:40:16,082 - INFO - Loaded plugin MirrorTool
2016-08-19 18:40:16,086 - INFO - Loaded plugin CameraTool
2016-08-19 18:40:16,095 - INFO - Loaded plugin ScaleTool
2016-08-19 18:40:16,095 - WARNING - Plugin ConsoleLogger was already loaded
2016-08-19 18:40:16,099 - INFO - Loaded plugin STLReader
2016-08-19 18:40:16,102 - INFO - Loaded plugin OBJReader
2016-08-19 18:40:16,106 - INFO - Loaded plugin STLWriter
2016-08-19 18:40:16,109 - INFO - Loaded plugin OBJWriter
2016-08-19 18:40:16,117 - INFO - Loaded plugin 3MFWriter
2016-08-19 18:40:16,120 - INFO - Loaded plugin WireframeView
2016-08-19 18:40:16,123 - INFO - Loaded plugin SimpleView
2016-08-19 18:40:16,127 - INFO - Loaded plugin FileLogger
2016-08-19 18:40:16,134 - INFO - Loaded plugin LocalFileOutputDevice
2016-08-19 18:40:16,139 - INFO - Loaded plugin XRayView
2016-08-19 18:40:16,143 - INFO - Loaded plugin AutoSave
2016-08-19 18:40:16,143 - WARNING - Plugin CuraEngineBackend was already loaded
2016-08-19 18:40:16,153 - ERROR - Exception starting plugin Doodle3D: NotImplementedError('Start should be implemented by subclasses',)
2016-08-19 18:40:16,153 - INFO - Loaded plugin Doodle3D
2016-08-19 18:40:16,158 - INFO - Loaded plugin GCodeWriter
2016-08-19 18:40:16,165 - INFO - Loaded plugin LayerView
2016-08-19 18:40:16,172 - INFO - Loaded plugin ImageReader
2016-08-19 18:40:16,178 - INFO - Loaded plugin SliceInfoPlugin
2016-08-19 18:40:16,183 - INFO - Loaded plugin LegacyProfileReader
2016-08-19 18:40:16,189 - INFO - Loaded plugin PerObjectSettingsTool
2016-08-19 18:40:16,194 - INFO - Loaded plugin SolidView
2016-08-19 18:40:16,201 - INFO - Loaded plugin PostProcessingPlugin
2016-08-19 18:40:16,205 - INFO - Loaded plugin CuraProfileReader
2016-08-19 18:40:16,225 - INFO - Loaded plugin USBPrinting
2016-08-19 18:40:16,232 - INFO - Loaded plugin ChangeLogPlugin
2016-08-19 18:40:16,238 - INFO - Loaded plugin GCodeProfileReader
2016-08-19 18:40:16,244 - INFO - Loaded plugin CuraProfileWriter
2016-08-19 18:40:16,252 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-08-19 18:40:16,258 - INFO - Loaded plugin 3MFReader
2016-08-19 18:40:16,259 - INFO - Command line arguments: {'file': [], 'disable-textures': False, 'debug-mode': False, 'external-backend': False}
2016-08-19 18:40:16,466 - WARNING - No active machine found when trying to save setting visibility
2016-08-19 18:40:16,474 - DEBUG - Attempting to kill the engine process
2016-08-19 18:40:16,701 - INFO - Started engine process: /opt/cura/bin/CuraEngine
2016-08-19 18:40:16,708 - DEBUG - Setting active view to SolidView
2016-08-19 18:40:16,739 - DEBUG - Backend connected on port 49674
2016-08-19 18:40:16,904 - DEBUG - Calculating normals took 0.00054931640625 seconds
2016-08-19 18:40:16,905 - DEBUG - Loaded a mesh with 8052 vertices
2016-08-19 18:40:17,369 - DEBUG - Loaded theme /opt/cura/share/cura/resources/themes/cura
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
2016-08-19 18:40:18,270 - DEBUG - Initialized OpenGL subsystems.
2016-08-19 18:40:18,270 - DEBUG - OpenGL Version:  4.4.0 NVIDIA 340.96
2016-08-19 18:40:18,270 - DEBUG - OpenGL Vendor:   NVIDIA Corporation
2016-08-19 18:40:18,271 - DEBUG - OpenGL Renderer: GeForce GTX 460/PCIe/SSE2
file:///opt/cura/lib/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///opt/cura/lib/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///opt/cura/lib/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///opt/cura/lib/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///opt/cura/lib/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///opt/cura/lib/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///opt/cura/lib/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
2016-08-19 18:40:23,159 - DEBUG - Calculating normals took 0.006087541580200195 seconds
2016-08-19 18:40:23,159 - DEBUG - Loaded a mesh with 112290 vertices
2016-08-19 18:40:23,190 - DEBUG - Loading mesh took 0.5383222103118896 seconds
2016-08-19 18:40:23,881 - DEBUG - Attempting to kill the engine process
2016-08-19 18:40:23,881 - DEBUG - Killing engine process
2016-08-19 18:40:23,882 - DEBUG - Engine process is killed. Recieved return code -15
2016-08-19 18:40:24,514 - INFO - Backend crashed or closed. Restarting...
[... 18 lines cut ...]
2016-08-19 18:40:33,027 - DEBUG - Killing engine process
2016-08-19 18:40:33,028 - DEBUG - Engine process is killed. Recieved return code -15
2016-08-19 18:40:33,766 - INFO - Backend crashed or closed. Restarting...
2016-08-19 18:40:33,869 - DEBUG - Attempting to kill the engine process
2016-08-19 18:40:33,870 - DEBUG - Autosaving preferences, instances and profiles
2016-08-19 18:40:33,887 - INFO - Started engine process: /opt/cura/bin/CuraEngine
[... cut ...]
2016-08-19 18:40:50,655 - INFO - Backend crashed or closed. Restarting...
2016-08-19 18:40:50,758 - DEBUG - Attempting to kill the engine process
2016-08-19 18:40:50,772 - INFO - Started engine process: /opt/cura/bin/CuraEngine
2016-08-19 18:40:50,797 - DEBUG - Backend connected on port 49674
file:///opt/cura/lib/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:151: TypeError: Cannot read property of null
file:///opt/cura/lib/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:151: TypeError: Cannot read property of null

opensourcefan, 2016-08-20:
Post Crash Report:

BLAS : Program is Terminated. Because you tried to allocate too many memory regions.
/usr/bin/cura: line 3: 21347 Segmentation fault      (core dumped) LD_LIBRARY_PATH=/opt/cura/lib PYTHONPATH=/opt/cura/lib/python3/dist-packages QT_PLUGIN_PATH=/opt/cura/lib/plugins QML2_IMPORT_PATH=/opt/cura/lib/qml QT_QPA_FONTDIR=/opt/cura/lib/fonts python3.4 /opt/cura/bin/cura "$@"

daid, 2016-08-20:
You are running out of memory.

closed 2016-08-20 by daid

opensourcefan, 2016-08-20:
Ya, gathered that. Any idea how or why 16gb of RAM wouldn't be sufficient?

daid, 2016-08-20:
Running out of address space?

BagelOrb, 2016-08-31:
Please log this issue at https://github.com/Ultimaker/Cura/issues
It seems you are reporting an issue of Cura 2.x at the repo for Cura 15.x

opensourcefan, 2016-09-02:
Yes, I apologize, I realized that a few days after. Didn't know that the newest version was a re-write. I'll do some more testing.