daid/LegacyCura #1533: Feature request: Support multiple extruder setups better (3+)

kyrreaa, 2016-08-23:
My 3x feed setup has 16 virtual extruders for color blending.
It requires a purge to get a clean new color and I was wondering if that will be supported soon by Cura?
Right now it tries to deposit 16 layers on layer 0 of the prime column which obviously fails miserably.

Ideally it could build a prime tower that is sparser when little material needs to be deposited, and denser if multiple changes are required during a layer. This way there is room for the material, and no extra material is put down just to lift the tower to next layer. It could simply bridge the next sparse layer ontop of the last, and fill it in with multiple primes if needed.

nallath, 2016-08-24:
We're still working on getting 2 extruders to work correctly. Getting 16 (!!) to work is a whole different beast. 

We also don't have any machines that support this, so don't expect a whole lot of work from our end anytime soon.

kyrreaa, 2016-08-24:
I understand.
Right now I got it to work quite well with purge box on the side and a silicone wiper.
I had to do this as the prime tower didn't work on multiples hehe.

But, do you get my idea though?
It appears to work with dual now as I mentioned, but simply increasing the purge volume for each nozzle is enough to allow multi setups to work. The fill would then be accross the purge column but for a 3 nozzle setup it would be filled by every other line. The other space is then left for the other nozzles purge.
If no nozzle has to purge, the active nozzle is used and a sparse layer is put down based on some bridging or fill factor.

I'll try and write a post processor that does this first, so I can see how it functions.
Maybe that makes it easier to add later?

Ghostkeeper, 2016-08-25:
Actually, the engine supports up to 16 extruders right now. And that 16 is just the size of a fixed-length array.

I think it would be a good idea to make the prime tower sparser in multi-extrusion set-ups. It's not really a problem for dual extrusion, but I can imagine it getting a bit of a problem when you have 16.

The prime tower is just filled with lines-infill as far as I know. The same algorithm as for infill, skins and support. There is already functionality there to set a different line distance, which we use to achieve the desired infill density. It should be fairly easy to fix. I'll track an issue for you in our internal issue tracker, so that it gets put on the road map, but as you can imagine it won't have a whole lot of priority for now, with the pressure on us to release what's now being called 2.3.

A post processor won't really make it easier for us to implement this fix, though it might clarify exactly what you mean.