daid/LegacyCura #1536: Not succeeding in connecting the computer and the printer

jfmonahan, 2016-09-01:
The software seemed to load fine, but when everything is set up, I do not get the message that I succeeded in connecting to the computer.  Windows 10.

BagelOrb, 2016-09-01:
What type of printer do you have?

jfmonahan, 2016-09-01:
It's a Hic-Top (RepRap) Prusa Mendel I3.  It's a kit that I just assembled.

BagelOrb, 2016-09-02:
Please get an Ultimaker and see if the problem still exists. ^^

Also you might want to try the new Cura: Cura 2.x!
Download at software.ultimaker.com

jfmonahan, 2016-09-03:
Thanks.  Sadly, the new Cura doesn’t support the Mendel.  I did find a Win 10 Driver though.  Maybe I’ll get an Ultimaker some time, but this is my first machine and I wanted to put it together myself to learn how it works.

Ghostkeeper, 2016-09-05:
We've had trouble connecting via USB on Windows 10 because the Arduino drivers can't be installed on Windows 10. Windows 10 provides its own drivers and doesn't allow installing anything on top of that. You're probably running into that on Legacy Cura too.

There are some instructions here that you could try: https://ultimaker.com/en/resources/18627-change-arduino-drivers

jfmonahan, 2016-09-05:
Thank you very much.  In all my messing around, I actually got it to work.  As often happens when trying to make something work in Windows, I don't actually know what I did that worked.  If I did, I'd pass it on.  Oh, well, at least I know it can be done.