daid/LegacyCura #1537: Cura no longer inserting "autolevel" command

ArizonaClark, 2016-09-04:
It finally dawned on me that when I printed something that had been sliced with an older version of Cura, it was going through autoleveling but when I "resliced" the same object (to try a higher fill), the printer was NOT doing an autolevelling.  This is leading to the Z being nearly 1mm too low.  Using "more" on the earlier file (claims a slice date of 21-08-2015) and the latest one, I discovered that the following was missing:

G29        ;Run the auto bed leveling

Using a text editor I manually added it to the file sliced today, and it printed a lot better.

Printer is a Printrbot Model 1405 "Simple Makers Kit", controlled by a Raspberry Pi running an old version of Octoprint.

The Aug. 2015 slicing was done on a Windows XP machine, with a version of Cura installed around Mar. 2015.  Today's attempt was with 15.04.6 (with auto-update enabled).

I spent some time trying to find a setting in Cura to enable autoleveling, but did not find one.

ArizonaClark, 2016-09-04:
Realized I forgot to mention that today's attempt was running Cura under OS X El Capitan Version 10.11.6 on a MacBook Pro.

Ghostkeeper, 2016-09-05:
Fixed in Cura 2.3: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/printrbot_simple.def.json#L37

(It has a G29 command in the start g-code.)

BagelOrb, 2016-09-05:
> Fixed in Cura 2.3: https://github.com/Ultimaker/
> Cura/blob/master/resources/definitions/printrbot_simple.def.json#L37
>
> (It has a G29 command in the start g-code.)

Download Cura 2.3 beta at software.ultimaker.com