daid/LegacyCura #1542: Process not quitting after app closed

mistertilapia, 2016-10-21:
In Cura 15.04.6 on Mac, quitting the app does not terminate its process. In activity monitor, the Cura process runs long after the app has been closed and often eats up around 90% of the CPU. The only way to end this process is to enter Activity Monitor and terminate "Cura". In more unfortunate circumstances, the process will eat battery life unbeknownst to the user and cause the computer to unnecessarily heat up.

BagelOrb, 2016-10-21:
Try Cura 2.3!

daid, 2016-10-21:
15.04 had this problem forever, it's related to wxWidgets, and it is one of the many reasons we started to work on Cura 2

Ghostkeeper, 2016-10-24:
Perhaps we can help getting the printer definition transferred to Cura 2? It's going to need some testing from your end, though.

mistertilapia, 2016-10-25:
I'd love to test! Let me know how I can help.

BagelOrb, 2016-10-25:
> But Cura 2 doesn't support my printer 😢

I just converted all machine profiles to cura 2.
Took me a couple of hours more than I thought ':)

What machine did you have?

mistertilapia, 2016-10-25:
Oh wow, thanks for porting all that.
My machine is the Velleman Vertex K8400. Here's what was in my Cura 15 machine settings:
Bed size: 180 x 200 x 190
Gcode flavor: Reprap (Marlin/Sprinter)
Baud: 250000
Perhaps I can take some of the work off of your hands if adding a config is a C++ task, just let me know.

BagelOrb, 2016-10-27:
I didn't port your setup to cura 2.4. You will need to create a
yourmachine.def.json file and submit it to the ultimaker/cura repo via a
pull request if you want your printer to stay supported...