daid/LegacyCura #1546: Won't print

mrcrider, 2016-11-30:
I have my printrbot hooked up, Ive loaded a file and it won't print.

Ghostkeeper, 2016-11-30:
It won't start printing until you somehow get the g-code onto your Printrbot.

I suggest you save the g-code on an SD card and put the SD card into your Printrbot. This is the [preferred method](https://printrbot.zendesk.com/hc/en-us/articles/201730880-Can-I-print-directly-from-an-SD-card-without-a-computer-) according to Printrbot.

If you intend to print via the USB cable and you're on Windows, you must first download their USB drivers. See the link they post here: https://printrbot.com/project/simple-metal/

mrcrider, 2016-12-01:
Thanks for your response.
I have printed on this printer before so my computer has the drivers and I
can see the model in the Cura software so the g code is there.
Still working on it.
thanks

lfleitas, 2017-02-01:
I am having the exact same issue.  Have printed with no problem prior to now and it will not switch to print.  It stays in save.  Did yours ever get fixed?