daid/LegacyCura #1547: Cancel plugins

Jeroendewijs, 2016-12-16:
Accidentally I added plugin events in Cura. 
First I could not find a way to remove these events. 
It appeared that the remove button was hidden at the right side of the column, if by default the colomn is set too narrow. you need to widen the colomn to see this button. For first time users this could be frustrating.
So move this button to the left, at the right side of the question mark ;-) Not much work I guess.

BagelOrb, 2016-12-16:
If this is about Cura 1.15, then I should inform you that we are not developing that version any more.
If this is about Cura 2.x, please supply an issue in the corresponding repository: github.com/Ultimaker/Cura

Jeroendewijs, 2016-12-16:
Thanks, I didn't know version 15 was end of life. I downloaded 2.x, looks good and saw the remove option.

closed 2016-12-16 by Jeroendewijs