daid/LegacyCura #1549: Can't open big file

baistavri, 2017-01-02:
Hi All,

I have following problem (Cura 15.04) I'm trying to open big file of 3D of statue build with Zbrush. Size of STL is ~348mb , program is starting process to open it (mouse pointer changing to pc thinking :) ) and after few second there is only small zoom effect and nothing was loaded (process of opening completed) - there is nothing on screen. I have same issue when trying to open obj variant of file ~231mb. I'm opening these files in Rhinoceros or Solidworks w/o any problem. Any ideas what is wrong?

Ghostkeeper, 2017-01-03:
It's... running out of memory? As Cura was based on Python, memory use tends to be somewhat inefficient.

BagelOrb, 2017-01-03:
Try Cura 2.x!
Cura 1.15 is not being developed any more.

daid, 2017-01-04:
Yes, the UI is running out of memory with a file that size. Cura 15.04 is not designed to work with those large files.

Cura 2 should work better on this aspect, because it's a true 64bit application. Or you need to lower the polygon count in a design application.

closed 2017-01-04 by daid