daid/LegacyCura #1550: extension dans cura

arnaud45, 2017-01-08:
Bonjour, j'ai activé par erreur les extensions "pause at height" et "tweak at Z 4.0.2" et je n'arrive pas à les enlever. Si quelqu'un sait comment faire merci.

BagelOrb, 2017-01-08:
Nous be parlons Fransais tres bien, mais peut-être je paut aider.

Quelled version de Cura aver vous? I'll ny a pas de travaille a Cura 1.15
en plus. Tous notre efforts sont pour Cura 2.4.

La version qui vous peut downloader a software.ultimaker.com as aussi les
extensions "pause at height" et "tweak at Z 4.0.2". Cette extensions sont
reimplementé, donc c'est possible que les erreurs sont disparu.

Bon chance avec Cura 2.x!

Ghostkeeper, 2017-01-22:
En dessous de "Enabled plugins", l'extension Pause at height et Tweak At Z aura un bouton [X]. Cliquez sur le cien pour les enlever.